cspang1/jcap #15: Can't set tile dimensions which don't wholly fit into pixel dimensions

cspang1, 2017-10-31:
Explained here: http://forums.parallax.com/discussion/167603/displaying-nxm-pixel-tiles-when-not-divisible-by-tile-scanline-height#latest

konimaru, 2017-11-02:
> I'm following along in the the HYDRA manual and the discussion explicitly mentions a 640x480 video resolution @ 10x12 tiles

I had a look but I can't find that particular statement. The only time 10x12 is mentioned is in the tile map for the NTSC stuff.

Anyway, you can only map pixels to an integer number of scan lines (unless you want to introduce interpolation).

cspang1, 2017-11-02:
@konimaru I'll have to reread the manual but now that you mention it I do recall that being in the NTSC discussion. 

I'm in the process now of researching interpolation of video signals. There's no way to avoid having uneven duplication of some tile lines, and if the upscaling is aggressive enough it'll be very obvious and unattractive.

Thanks for the input!

closed 2017-11-02 by cspang1

konimaru, 2017-11-03:
When you're not tied to that tile resolution (10x12) then there are lots of suitable VGA drivers available. I think 256x192 could work for you (16x12 visible tile resolution).

cspang1, 2017-11-03:
@konimaru I actually just more or less completed my vga driver w/ adjustable screen tile dimensions and tile pixel dimensions, w/ some scrolling control code hard-coded for testing. If you're interested: https://github.com/cspang1/JCAP/tree/test/video/vga/8bitscroll/Dev/Video/VGA/Software