cspang1/jcap #19: Need to refactor VGA driver code to iteratively assign internal registers from main RAM

cspang1, 2017-11-06:
Currently, all registers in the driver cog which need to be populated with values calculated in the hub are done so one at a time in what's essentially an unrolled loop. Need to determine if there is a way to set them all in a single tight loop.

konimaru, 2017-11-06:
_It's a shame really. Those things have been figured out a long time ago ..._

Check out [this link](http://forums.parallax.com/discussion/comment/1108886/#Comment_1108886), it's a generic loop transferring **a long per hub cycle** in either direction. There is also a counter based version pre-dating this one which - while OK for me - wasn't generally accepted (and sometimes you don't have a spare counter). Anyway, use (one of) them if you want/need speed ...

konimaru, 2017-11-06:
I apologize for the mess Parallax made during forum transition(s) which is the reason I stopped participating.

cspang1, 2017-11-06:
@konimaru Thanks, looks like just the thing I need. Hadn't even gotten around to Googling for a solution yet so thanks for saving me a step! Only thing is that the procedure looks almost as nasty (instruction-wise and visually) as explicitly setting variables, no matter how you do it.

konimaru, 2017-11-06:
Your call. I - obviously - use this in most of my drivers (where applicable). And the setup is only required once.

cspang1, 2017-11-06:
@konimaru True story! I'm definitely riding the assembly-timing learning curve into outer space. I'm actually going to have to do a refactor of my VGA driver to populate a line buffer during the visible porch/sync regions, instead of racing to grab a new tile every time I need it. As-is the driver works fine up to 15 horizontal 16x16 tiles, or 16 horizontal 8x8 tiles. Past that, de-sync.

cspang1, 2017-11-07:
Satisfied by commit a2ff0f2167f0a29744ab3d61e357886df1cfc68d

closed 2017-11-07 by cspang1

konimaru, 2017-11-07:
For the record, I just realised you only wanted a one-off load not something repeating (where speed might be important).

cspang1, 2017-11-07:
@konimaru no problem! There's an entirely realistic chance the driver will change in the future to make that necessary, and being aware of the execution-time conscious approach is useful.