cspang1/jcap #20: Investigate multi-cog vs dedicated PPU video generation approach

cspang1, 2017-11-08:
Current driver has to render faster than it can load the next tile/color at high tile dimensions, and adding the sprite rendering feature will only exacerbate this. The benefits of spreading the rendering of each line over multiple cogs vs having a dedicated PPU with screen buffer needs to be investigated.

cspang1, 2017-11-10:
@konimaru I'm definitely interested in discussing what you discuss in this thread if you have the time: http://forums.parallax.com/discussion/126874

konimaru, 2017-11-10:
Ah, that one. I felt a bit odd replying to myself :) Anyway, what are you after?

cspang1, 2017-11-10:
I guess first off, what's the final "verdict" you reached on fully deterministic and synced **waitvid** timing? I see you go through quite a few iterations of solutions, and in your final version you freeze the video hardware before changing **vscl**. Does that have to be done for every change of **vscl** (for example changing for vertical blanking lines and HSync) or just initially?

konimaru, 2017-11-10:
Let me ponder this for a while. This is really **old** stuff and these days I simply _use_ it.

cspang1, 2017-11-10:
@konimaru Haha understood, no problem! I'm working on getting the general structure of this refactor together right now, so no rush!

cspang1, 2017-11-15:
@konimaru I've managed to implement a cog sync method (based on your and Linus's legwork) that's at least as good as it needs to be. I'm not going down the rabbit hole of WHOP methodology at this stage, and likely won't in the future, as the resolutions I'm working with don't require that kind of time-shaving and code obfuscation. I'm still interested in the odd behavior exhibited by the video generation hardware.

As far as this issue goes, I've chosen to go with a multi-cog approach over one requiring a dedicated PPU. My graphics demands simply aren't high enough to warrant a PPU, and at this point I've managed to wrap my head around the multi-cog approach enough to implement it.

closed 2017-11-15 by cspang1

cspang1, 2018-01-29:
This paradigm ended up being chosen as the way forward.