cspang1/jcap #23: Graphics driver can not be started again after being stopped

cspang1, 2017-11-11:
Calling graphics.start after graphics.stop (or calling graphics.start while driver is already running) results in no video.

konimaru, 2017-11-13:
What is this driver you speak of? The only [**graphics.spin**](https://github.com/cspang1/JCAP/tree/4a9e0d7ffa2518b786a3eae2e0d4fc0a2039d16b/Dev/Game/Software) I could find doesn't even have a **stop** method. And simply calling **start** on that one again will also start another VGA driver underneath (with expected results).

cspang1, 2017-11-13:
@konimaru it's on the feature/video/vga/multicog branch: https://github.com/cspang1/JCAP/tree/feat/video/vga/multicog/Dev/Game/Software

konimaru, 2017-11-13:
This should do the trick:
```
PUB start(graphics_addr_base) : vidstatus | nCogs,i                             ' Function to start VGA driver with pointer to Main RAM variables
    stop                                                                        ' Stop running driver
    repeat i from 0 to numCogs-1                                                ' Loop through cogs
      ifnot cog_[i] := cognew(@vga, graphics_addr_base) + 1                     ' Initialize cog running "vga" routine with reference to start of variable registers
        stop                                                                    ' Stop all cogs if insufficient number available
        abort FALSE                                                             ' Abort returning FALSE
    waitcnt($8000 + cnt)                                                        ' Wait for cogs to finish initializing 
    return TRUE                                                                 ' Return TRUE                                                
```
Note the extra **stop** call. This makes sure consecutive **start** calls don't attempt to start thousands of drivers. That said, this wouldn't explain the not-working **stop/start** sequence.

cspang1, 2017-11-13:
@konimaru That definitely did the trick, because in my test code for stopping and starting the driver I don't check if it's already running or not and the input to start/stop is constant, so it just spams start or stop...

`  started := false
  
  repeat
    if (control_state & 8) AND started
      started := false
      graphics.stop
    elseif (control_state & 4) AND !started
      started := true
      graphics.start` 

I code good.

Satisfied by commit 713ed567d48169d914dfebaab7385abe59c118cb

konimaru, 2017-11-14:
Out of curiosity, why did you put this logic into to the application? Isn't the driver itself better suited for this (i.e. keeping state info)?

cspang1, 2017-11-14:
@konimaru Yes it definitely is! I was just testing to make sure the start and stop routines worked, and the easiest way at the time was to control them with button presses. I forsee no real reason I should ever have to stop or restart the video driver unless I implement some sort of software reset for the whole system, so just novelty mixed with making sure everything is tested instead of assumed-working.

konimaru, 2017-11-14:
OK, I thought this is for deployment. Personally I only implement **init** methods. No-one is stopping things anyway so there :)

cspang1, 2017-11-14:
@konimaru That's definitely a cleaner design paradigm haha. 

Just a heads up, any branch that's preceded by "wip" is baseline "work in progress" (standard dev, such as for wip/video/vga or wip/video/sound), "feat" is "feature" (a unique dev aspect, such as feat/video/vga/multicog which is focused on refactoring my single-cog driver into a synchronized parallelized one). Other prefixes may be used for things like hotfixes etc. Nothing is merged with master until development on it is considered "complete" (for example, the wip/input branch was merged a while back because I considered the 74HC165 interfacing complete).The idea is that the only production/deployment code will be in master, and everything else is dev.

cspang1, 2017-11-17:
Starting/stopping driver exhibiting strange behavior again.

cspang1, 2017-11-27:
Stopping then starting the driver results in apparently valid sync waveforms, but not video.

cspang1, 2018-01-22:
Made unecessary by dual-Propeller development.

closed 2018-01-22 by cspang1