cspang1/jcap #27: suggestion re: data collection

konimaru, 2017-11-22:
While probably not terribly important right now I'd suggest a minor change to this:
```
        shl             ti,     tOffset         ' Multiply tile index by size of tile map
        add             ti,     tpbase          ' Increment tile index to correct line
        add             ti,     tpptr           ' Add tile palette pointer to tile index to specify row of tile to be displayed
rdtile  rdlong          tile,   ti              ' Read 16-pixel-wide tile from Main RAM
        shl             ci,     #2              ' Multiply color index by size of color palette
        add             ci,     cpbase          ' Increment color index to correct palette
        rdlong          colors, ci              ' Read tile from Main RAM
movp    mov             0-0,    tile            ' Store tile row to pixel buffer        
movc    mov             0-0,    colors          ' Store color palette to color buffer
        add             movp,   incDest         ' Increment tile buffer pointer
        add             movc,   incDest         ' Increment color buffer pointer
```
Here **tile** and **colors** are loaded from hub _then_ transferred to a cog location (2 steps). This could easily be done in a single step e.g.
```
        shl             ti,     tOffset         ' Multiply tile index by size of tile map
        add             ti,     tpbase          ' Increment tile index to correct line
        add             ti,     tpptr           ' Add tile palette pointer to tile index to specify row of tile to be displayed
rdtile
movp    rdlong          0-0,    ti              ' Read 16-pixel-wide tile from Main RAM
        shl             ci,     #2              ' Multiply color index by size of color palette
        add             ci,     cpbase          ' Increment color index to correct palette
movc    rdlong          0-0,    ci              ' Read tile from Main RAM
        add             movp,   incDest         ' Increment tile buffer pointer
        add             movc,   incDest         ' Increment color buffer pointer
```

konimaru, 2017-11-22:
... and to bring this loop a bit (more) in sync with hub window slots:
```
line    rdbyte          ti,     tmptr           ' Read tile index of current map tile
        add             tmptr,  #1              ' Increment tile map pointer to color index
        shl             ti,     tOffset         ' Multiply tile index by size of tile map
        rdbyte          ci,     tmptr           ' Store color index of current map tile
        add             ti,     tpbase          ' Increment tile index to correct line
        add             ti,     tpptr           ' Add tile palette pointer to tile index to specify row of tile to be displayed
rdtile
movp    rdlong          0-0,    ti              ' Read 16-pixel-wide tile from Main RAM
        shl             ci,     #2              ' Multiply color index by size of color palette
        add             ci,     cpbase          ' Increment color index to correct palette
movc    rdlong          0-0,    ci              ' Read tile from Main RAM
        add             movp,   incDest         ' Increment tile buffer pointer
        add             movc,   incDest         ' Increment color buffer pointer
        add             tmptr,  #1              ' Increment tile map pointer to next tile in row
        djnz            tptr,   #line           ' Generate one scanline of data
```

cspang1, 2017-11-22:
@konimaru Great suggestions. A refactor of the tile/color loading to minimize and hub access sync the routine was definitely in order.

cspang1, 2017-11-22:
Satisfied by commit e3f658b90084b0c7b4d8ea538a7e07b9c4d766e0

closed 2017-11-22 by cspang1