cspang1/jcap #29: Refactor graphics and VGA interaction to clean up parameters

cspang1, 2017-11-27:
Right now [the VGA driver instantiation from the graphics system](https://github.com/cspang1/JCAP/blob/wip/video/vga/Dev/Game/Software/graphics.spin#L93) is trash; passing completely random parameters like VSync attributes and tile line to scanline ratios. Solution is to pass a pointer to the base of them and perform a **longmove**. Only parameters passed then will be a base pointer to the graphics resources and a base pointer to cog initialization attributes. Consideration needs to be made about exactly what should be calculated in the Graphics vs VGA object. A lot of the calculations will overlap between VGA and RGBS, so it would be poor design to calculate the same attributes separately in the individual video drivers. 

Some attributes are used directly by the video driver cog(s) via **rdlong**s, while others are set just before cog instantiation because they're different depending on the cog (e.g. different vertical porch sizes for interlacing). Theoretically you could set ALL attributes in Spin before cog initialization, which would free up a bunch of space currently used to **res** and **long** variables, as well as remove the need to copy memory from the hub to the cog. This however greatly obfuscates the code. Right now this is being done in [a couple places it doesn't need to be](https://github.com/cspang1/JCAP/blob/wip/video/vga/Dev/Game/Software/vga.spin#L102). 

Ultimately too much high-level code styling is being considered as the real core of the code is low-level. The added consideration of cog code size is a driving factor.

konimaru, 2017-11-28:
> A lot of the calculations will overlap between VGA and RGBS, so it would be poor design to calculate the same attributes separately in the individual video drivers.

Then again, too much generalisation kills you in the end.

I would also consider whether it's worth doing certain calculations live or **constant** based (compile time). If I were to configure a certain (game) setup I wouldn't expect user input wrt sprite resolution. IOW certain parameters never change (again).

cspang1, 2017-11-28:
It would be the choice between a library of various drivers with different resolutions and tile dimensions and etc. or a single one or two which allows a developer to set those at runtime. Obviously pros and cons of each, but the former has the benefit of reducing the driver memory footprint and being more true to how retro hardware would itself necessarily be constrained.

cspang1, 2018-01-22:
Made unecessary by dual-Propeller development.

closed 2018-01-22 by cspang1