cspang1/jcap #30: [request] compatibility test

konimaru, 2017-11-28:
Seeing that you seem to have P1V hardware available (FPGA), could you please test the following fragment and report the result?
```
                org     0

test            jmpret  $, #setup

' light up an LED here or have some indication that the board is running.

                waitpeq $, #0

                hubop   vscl, #%10000_000
setup           mov     vscl, $-1
                jmp     #$1FF
```
Expected result (based on P1 evidence) is that the **running** indicator should be active.

The P1 has the odd quirk that instructions at address $1FF are cancelled. So in the program above, the REBOOT placed there should be ignored and execution should continue at address 0 falling through to 1 where e.g. an LED is lit followed by an endless **wait**(**peq**).

OTOH, if the REBOOT does fire, the **running** indicator won't be reached.

cspang1, 2017-11-28:
Sure thing! I'll jump on it when I get home.

konimaru, 2017-11-28:
Thanks!

> The P1 has the odd quirk that instructions at address $1FF are cancelled.

Which doesn't mean that it can't be done :)

cspang1, 2017-11-28:
@konimaru So the behavior you described is verified. The following code was used:

```
CON
        _clkmode = xtal1 + pll16x                                               'Standard clock mode * crystal frequency = 80 MHz
        _xinfreq = 5_000_000

PUB main
  cognew(@tester,0)

DAT
              org     0

tester  jmpret          $,      #setup

        or              dira,   Pin_outs        ' Set output pins
        or              outa,   Pin_CP          ' Drive clock pin high

        waitpeq         $,      #0

        hubop           vscl,   #%10000_000
setup   mov             vscl,   $-1
        jmp             #$1FF        

Pin_outs      long      |< 0                    ' 74HC165 clock pin bitmask
Pin_CP        long      |< 0                    ' 74HC165 clock pin bitmask        
```

I copied the pin-driving lines directly from my 74HC165 driver, and used the logic analyzer test bench for the driver to verify:

[image]

Pin 0 is driven high, so the **running** indicator was reached.

closed 2017-11-28 by cspang1