cspang1/jcap #31: Develop dual-Propeller "display" routine

cspang1, 2018-01-22:
https://github.com/cspang1/JCAP/projects/1#card-6846453

cspang1, 2018-01-27:
Satisfied by commit afe2b99b6fc627876033ce76f14d877c2d639d02

closed 2018-01-27 by cspang1

konimaru, 2018-01-29:
Just an observation. You issue **4 pixels** with each **rdlong/waitvid** pair. The worst case time for this sequence is 30 cycles (23+7). At 104MHz (and 25.175MHz pixel clock) that's **7.26 pixels** _you have to_ deliver.

cspang1, 2018-01-29:
I'm not exactly following. I'm rendering 4 pixels per waitvid but they're upscaled via **vscl** to 8 pixels. I've recalculated my **frqa** with 104 MHz and replaced [the old value](https://github.com/cspang1/JCAP/blob/bd81e2bbe446af62310c9bd30a0cfd616fe53224/Dev/Game/Software/vga_display.spin#L93) with [the new](https://github.com/cspang1/JCAP/blob/wip/dualprop/render/Dev/Game/Software/vga_display.spin#L107).

I've also analyzed the video signal with an oscilloscope and the timings are [almost nanosecond perfect](http://tinyvga.com/vga-timing/640x480@60Hz).

Is there something I'm missing?

konimaru, 2018-01-30:
Sorry, my mistake. I was under the assumption that you go for full 640. With 320 that's perfectly OK. Somehow I missed the **vscl** setup. That said, 320 at 80MHz are possible.

cspang1, 2018-01-30:
Oh no problem haha, thanks for the oversight!

konimaru, 2018-01-30:
I'm getting old :)

Anyway, I'm glad you keep going (was a bit quiet over the holidays).

cspang1, 2018-01-30:
Yeah I was in limbo with the way forward on graphics. I was opposed to a dual-Propeller setup at first because I wasn't certain of its necessity, then I was opposed to running the Prop at 104 MHz because it seemed like a workaround instead of addressing directly the issue of reading and displaying longs back-to-back in time. But at the end of the day, two Props give me an outstanding amount of processing power and time to generate all of the graphics and contituent effects I want, and 104 MHz lets my display code be a lot simpler and gives me even _more_ time to generate graphics. Parallax has tested and approved that frequency, and so far my own testing has confirmed its stability. So now that I've gotten over myself I can push forward haha.