cspang1/jcap #34: lock usage is questionable

konimaru, 2018-01-30:
**vga_render.spin**
```
  long  var_addr_base_          ' Variable for pointer to base address of Main RAM variables
  byte  cog_sem_                ' Cog semaphore
  long  start_line_             ' Variable for start of cog line rendering
```
This being in a VAR section will actually place **cog_sem_** after **start_line_** (first all **long**s, then all **word**s followed by all **byte**s). In a DAT section the order remains but alignment is forced (3 unused bytes between **cog_sem_** and **start_line_**).

Anyway, I'd simply pick a **long** and that's this problem out of the way. Next issue is that you actually have to read the value, right now you use its assumed address, e.g.
```
  long  var_addr_base_          ' Variable for pointer to base address of Main RAM variables
  long  cog_sem_                ' Cog semaphore
  long  start_line_             ' Variable for start of cog line rendering

        rdlong          clptr,  par             ' Initialize pointer to current scanline
        add             semptr, par             ' Initialize pointer to semaphore
        add             ilptr,  par             ' Initialize pointer to initial scanline
        rdbyte          sem, semptr             ' Get semaphore ID

:lock   lockset         sem wc                  ' Attempt to lock semaphore

semptr  long    4       ' Pointer to location of semaphore in Main RAM
ilptr   long    8       ' Pointer to location of initial scanline in Main RAM

sem     res     1       ' semaphore (or reuse semptr)

```
ATM it works because all cogs have the same value in **semptr** (**par+1**) of which the lowest 3 bits are used as a lock ID (**%-01**, locks don't have to be checked out to be used).

cspang1, 2018-01-30:
Good catch! Wasn't aware of the `VAR` variable organizing behavior. I'll address this when I get home.

cspang1, 2018-01-30:
Satisfied by commit 9198014

konimaru, 2018-01-31:
Your are still using the pointer (where the index is stored) instead of the semaphore index (**lockset** works on a cog register and _doesn't_ do hub access).

cspang1, 2018-01-31:
Thought I'd included that! It's a bugger that the locks work in spite of improper implementation, otherwise without oversight like yours it would literally never be addressed, and any problems that would arise because of it would be incredibly difficult to track down. Thanks!

cspang1, 2018-01-31:
Re-satisfied by 449cb6c2159f40856ba4cd10d3785d4500e19272

closed 2018-01-31 by cspang1