cspang1/jcap #43: Need to further optimize render routine to hit 64-sprite SAT

cspang1, 2018-02-19:
Target size for the sprite attribute table (SAT) is 64 sprites, with a maximum of 8 sprites per scanline. 

Right now with the worst case scenario of sprites which have zero transparent pixels: I'm missing the scanline deadline at all 64 sprites and 8 sprites on a line. However, incremental testing has shown that I'm only missing by a factor of a sprite or two per line and ~10 sprites on the screen.

Execution time needs to be shaved off either in the tile rendering section (already unrolled loops) or the sprite visibility check area.

@konimaru I may need your magic on this one.

konimaru, 2018-02-22:
> [@konimaru](https://github.com/konimaru) I may need your magic on this one.

No problem. I'm on holiday right now. I call back :)

Cheers,
Marko

cspang1, 2018-02-22:
@konimaru Sound good; hope you're having a great time! 

The code in question is **vga_render.spin** on the **wip/video/sprites** branch.

I think if time can be shaved off anywhere it's in either the [sprite visibility check section](https://github.com/cspang1/JCAP/blob/3dffbcd24f771f3fa689c1434551eaeee7661267/Dev/Game/Software/vga_render.spin#L206) or the [sprite tile line parsing section](https://github.com/cspang1/JCAP/blob/3dffbcd24f771f3fa689c1434551eaeee7661267/Dev/Game/Software/vga_render.spin#L291).

Like I said above, bumping up the number of render cogs from 5 to 6 (thus overwriting the SPIN interpreter in Cog 0) lets me hit my goal as well as give me time for some extra effects like fine and parallax scrolling.

So unless you see a really critical optimization, to the tune of hundreds of instructions per line, I think the code is in its almost final state.

Thanks!

konimaru, 2018-03-07:
OK, I'm back to my daily grind. Is that **vga_render.spin** still current, i.e. can I work with/on it?

cspang1, 2018-03-07:
@konimaru welcome back! You'll want to look at HEAD~1 of wip/video/sprites for the last working version of render. In the most recent commit I started working on getting fine sprite scrolling down. Take a look here for specifics on that line of effort: https://forums.parallax.com/discussion/168164/shifting-consecutive-longs

If you look at the most recent commit, all the commented out code will certainly be removed in place of whatever algorithm I implement to populate the buffer w/ fine scrolling.

cspang1, 2018-06-07:
Tentatively satisfied by commit 7de0fe55c306b9192f03856ddd758dfb50977b88

closed 2018-06-07 by cspang1