cspang1/jcap #44: Implement fine tile map scrolling and sprite size indexing

cspang1, 2018-02-19:
Need to be able to smoothly scroll the tile map, and some sort of x/y sprite size indexing.

cspang1, 2018-02-23:
Sprite size indexing will likely not be implemented (too great a hit to sprite system render time).

cspang1, 2018-03-07:
https://forums.parallax.com/discussion/168164/shifting-consecutive-longs

konimaru, 2018-03-08:
Are you simply after pixel shifting as the last step before it goes to the screen or is this to happen earlier in the pipeline (with subsequent steps)? If it's the former you don't really have that much overhead as only the left/right ends need special attention. Everything in between is send to the screen as usual (even with aligned loads).

For example, in my [Gameduino style text mode driver](https://github.com/konimaru/waitvid.2048/tree/master/Gameduino/51xG%20RAM) I use pixel scrolling by simply having a render loop for each alignment.

cspang1, 2018-03-08:
@konimaru I guess originally I intended for scrolling to be done on the tiles only, independent of the sprites. 

What do you think? There are pros and cons for scrolling the tile map separately from the sprites depending on the game type. Obviously scrolling everything together is the "easiest" method.

I can't load the link you sent me for some reason (work network -.-), but when you say you have different render loops you mean the method of generating the actual scanline bitmap or **waitvid**ing it out to the screen?

konimaru, 2018-03-09:
It's simply the last step going from memory to video h/w, e.g. **rdlong -> waitvid**.

What was you original intention? Background independent from sprites?

cspang1, 2018-03-09:
Oh gotcha; I'll have to look at your code but I don't think I have the nanoseconds left to do anything else in my scancode. And not necessarily, just whatever works best.