cspang1/jcap #47: RX cog can miss TX ACK given worst-case hub access times

cspang1, 2018-05-16:
[image]

Fix is to add at least 3 **nop** instructions [here](https://github.com/cspang1/JCAP/blob/f5575c87971d57fab43bd27f4b189d217942d3e3/Dev/Game/Software/vga_tx.spin#L67).

konimaru, 2018-05-17:
Shouldn't the delay come **after** the **long** has been sent, or before if you like but definitely not in the middle (between start bit and data)?

cspang1, 2018-05-17:
@konimaru you're right haha, I fat fingered the permalink. **nop**s should be placed [here](https://github.com/cspang1/JCAP/blob/f5575c87971d57fab43bd27f4b189d217942d3e3/Dev/Game/Software/vga_tx.spin#L66).

cspang1, 2018-05-18:
Satisfied by commit 0e2d445767d28afc13386d538d89271c4bd5f282

closed 2018-05-18 by cspang1