cspang1/jcap #48: TX/RX Routine failing after first full buffer transfer

cspang1, 2018-05-16:
Discussed [here](https://github.com/cspang1/JCAP/commit/f5575c87971d57fab43bd27f4b189d217942d3e3#commitcomment-28964282)

cspang1, 2018-05-18:
Ok @konimaru tell me something: should changing a **dira** bit to 0 (input) not cause that pin to be driven low? Even when there's a pull-down resistor on that pin?

cspang1, 2018-05-18:
Satisfied by commit 0e2d445767d28afc13386d538d89271c4bd5f282

Question still remains of **dira** behavior.

closed 2018-05-18 by cspang1

cspang1, 2018-05-18:
@konimaru my entire project is now running on 100% actual hardware: two P8X32As w/ support circuitry et. EEPROMs. I'm using a 10K pull down resistor on the data line. Does this sounds like a fault uC IC then?

konimaru, 2018-05-18:
Let me try this at home. How long is your cable?

Can you easily change pins?

cspang1, 2018-05-18:
@konimaru cable is a few inches; the two uCs are on the same standard breadboard (breadboards being notorious for signal noise and ground plane sinking, maybe a source of the problem itself?) 

Pins are easily changed in both TX and RX: TX by changing the variable in the vga_tx CON section, and RX by changing the variable at the bottom of the vga_rx PASM section. I should probably standardize that, eh?

I've tried changing pins during debugging to no avail. 

I don't say it enough but thanks again for lending your time, effort and experience to my little hobby project :)

konimaru, 2018-05-18:
So effectively the sequence (plus pull-down)
```
or      outa, TxPin
or      dira, TxPin
andn    dira, TxPin
```
is all I need to see if it works or not?

cspang1, 2018-05-18:
@konimaru yep that should be the most trivial case

cspang1, 2018-05-18:
@konimaru I completely agree! I can do a sanity check when I get home, but I'm 99.9% certain one end of a 10K resistor is attached to pin 0, and the other end is attached to ground. Maybe I'm getting some strange impedence effect, and I need to lower the pull down resistor value?

konimaru, 2018-05-19:
Just checked on my trusty demoboard with a 10K pulldown. One cog just samples a pin (3) and copies its state to an LED pin (23). With the sequence in question (even in Spin) it stays high after the pin is set floating (no pulldown) for a while. With the pulldown in place I never see a blip on the LED.

konimaru, 2018-05-20:
At PASM level I get an expected 4 cycles with forced **outa** toggling. Switching off **dira** instead clocks in with 23 cycles (10K pull-down). So maybe the 10K are a bit high for this purpose.
```
CON
  _clkmode = XTAL1|PLL16X
  _xinfreq = 5_000_000

CON
  PIN = 3
  
VAR
  long  stack[32]
  
PUB null

  cognew(@main, 0)
  cognew(link, @stack{0})
{
  waitcnt(clkfreq + cnt)
  
  outa[PIN] := 1
  dira[PIN] := 0
} 
  waitpne(0, 0, 0)

PRI link

  ctra := constant(%0_11010_000 << 23 | PIN)            ' LOGIC A
  frqa := 1
  
  waitpeq(%1000, %1000, 0)
  waitpne(%1000, %1000, 0)
  
  dira[23..16]~~
  outa[23..16] := phsa <# 255

  waitpne(0, 0, 0)
  
DAT             org     0

main            or      dira, #|< PIN

                rdlong  cnt, #0
                add     cnt, cnt
                waitcnt cnt, #0

                or      outa, main
'               mov     outa, #0                        '  4 cycles
                mov     dira, #0                        ' 23 cycles (10K pull-down)

                waitpeq $, #0

                fit
                
DAT
```

cspang1, 2018-05-20:
@konimaru Interesting; I was under the impression that 10K was standard across the board as a pull-down value for P8X pins. I'm thinking this may also be the cause of issue #49. Thanks for the legwork!

konimaru, 2018-05-20:
I'd leave the pull-down in (just to have a base line) and do a forced **outa**.

cspang1, 2018-05-21:
@konimaru Yeah that's going to have to be the solution. I cut the resistor value in half and was _still_ not seeing the appropriate behavior. I'm starting to wonder whether this is a breadboard impedance issue, or something caused by connecting the two props. Because, logically, the only two explanations are:

1. Vss is floating high relative to Vdd, causing the appropriate differential to not be met
2. There is either low impedance on the Prop-Prop connection causing cross-talk that bypasses the pull-down or somehow the TX Prop is still driving a logic low at the critical moment

cspang1, 2018-05-22:
@konimaru scratch all of that I'm picking up what you're putting down now. So the time constant for the circuit created by the interconnected prop is defined by a 20 pF mutual capacitance. Just based on the 23 sysclock count you identified, I need to drop the resistor down to 1K (minimum of 1.4 to be precise). 

Regardless, I'm going to keep the explicit driver instructions because a digital margin of error is better than an analogue margin of error.