cspang1/jcap #54: Research implementing 8-bit color w/ separate VSYNC/HSYNC pins

cspang1, 2018-06-06:
Might be impossible due to already-tight _vga_display_ routine timing, but worth the four-fold increase in color fidelity.

konimaru, 2018-06-07:
Been there, done that, works :)

cspang1, 2018-06-07:
@konimaru Yep someone mentioned your name on the forum when discussing it!

Without looking at your implementation, I'm wondering how you were able to toggle the HSync/VSync outputs deterministically. 

e.g. **waitvid** will finish displaying the visible region of the scanline, then transition to the horizontal front porch. Pulling down the HSync pin via **andn _outa_, HSPin**  even the instruction immediately after the HSync region **waitvid** was executed would still introduce a data path's amount of lag into the toggling of the pin.

konimaru, 2018-06-07:
That's not quite how it works. While I do the visible stuff the sync lines are kept idle by **outa**. Basically, after the last pixel I register the hsync stuff normally with a **waitvid** (so timing is exact) and only then I switch byte lanes in **vcfg** to have the sync pulse on the correct pins. Just before the front porch runs out I switch lanes again and **outa** takes over again.

cspang1, 2018-06-07:
@konimaru ah I see, that makes perfect sense, thanks!

I can see how active-high sync signals would be trivial to implement with pull-down resistors, however my sync signals are active-low.

konimaru, 2018-06-08:
First, apologies, I got front/back porch mixed up (I was under the impression it's in **front** of the video signal when in fact it's in relation to the sync pulse).

Anyway, no messing around with **dira** here.
- **dira** covers 8bit video and 2bit sync from the start and is never touched again
- **outa** is set (initially) to idle state (high) in the sync pin location
- video line is displayed
- hsync **waitvid** is issued (front+sync+back)
- immediately after that **vcfg** is updated (pin group) then **outa** is cleared (no interference)
- now we have to calculate a point in time when the back porch becomes active (sync lines high again)
- at this point **outa** goes back to idle (take over) followed by a **vcfg** pin group switch for the next scan line

The last point has to be carefully adjusted because we don't want to bleed the (now high) idle state into the colour group before its time. Admittedly that is the case in my driver (for a few ns) but I like stuff to be clean.

konimaru, 2018-06-08:
FWIW, the bleeding can be avoided with an exotic sync pattern, i.e. keep low until switchover (**vcfg**) then high before release (**outa**) and reverse that before going back to visible video.

cspang1, 2018-06-08:
@konimaru No problem! If you google "vga porch timing" you'll see it BOTH ways all over the place; makes have a technical discussion about it without a reference much more difficult.

I was under the impression that **vcfg**, like **vscl**, is latched at the moment of the next **waitvid** execution.

konimaru, 2018-06-08:
In my experience **vcfg** takes immediate effect :) There was a QuickStart extension board for PropellerBASIC which had a 256colour mode. It wouldn't have worked since I use a single **waitvid** for hsync.

cspang1, 2018-06-08:
@konimaru Interesting, thanks! Really wish that kind of info was in the Prop datasheet!

I took a crack at an "exotic" sync pattern w/o color bleeding. See anything obviously wrong?

[image]

cspang1, 2018-06-11:
@konimaru No problem, being on opposite sides of the world has its inconveniences haha! Good call on the back porch as well; thinking about it this morning and I felt I was missing something.

For this project, transparency and maintainability of operation are definitely high priority, as a goal is for anyone to be able to come behind and modify the drivers if a game calls for it.

closed 2018-06-11 by cspang1