cspang1/jcap #62: Decouple TX/RX From VSYNC

cspang1, 2018-06-20:
Greatly over-complicates implementing anything time-sensitive.

cspang1, 2018-06-21:
@konimaru I'm strongly considering using [your high speed serial routine](http://forums.parallax.com/discussion/110325/fyi-single-wire-high-speed-serial-link-updated-20110514) depending on its transmission speed. I'm getting ~13 Mbps @ 104 MHz system clock currently.

cspang1, 2018-06-21:
@konimaru makes sense, thanks!

Potentially stupid question: how does the `wrlong  par, par` clear the `par` register to zero? Wouldn't `wrlong` take the value in `par` (size:addr) and just store it into whatever register is created by the combination of size:addr?

konimaru, 2018-06-22:
Yes and yes. These days I use `wrlong zero, par` with `zero` being a constant with value `$1F0`.

cspang1, 2018-06-25:
Alright @konimaru, I'm officially at a loss. [Calling the assembly via `cognew` works perfectly](https://github.com/cspang1/JCAP/blob/bug/txrx/refactor/Dev/Software/cpu.spin#L48); no issues.

But if I comment that out, and uncomment the line above which invokes [the exact same assembly except via an object](https://github.com/cspang1/JCAP/blob/bug/txrx/refactor/Dev/Software/tx.spin), the routine does nothing and I cannot get any debugging done deterministically.

I can only imagine it has _something_ to do with the address being passed it, but I can't fathom what it is. Thoughts?

konimaru, 2018-06-26:
[This line](https://github.com/cspang1/JCAP/blob/b5546437028c531eaaeb0d20672d0f42e10a7586/Dev/Software/tx.spin#L54) should read `:primary rdlong buff, tx_addr`.

cspang1, 2018-06-26:
*screams in pain internally*

cspang1, 2018-06-27:
Satisfied by commit 95ec6d062ec8657aaefdfb8121d3011ebde27ef6

Thanks again @konimaru!

closed 2018-06-27 by cspang1