cspang1/jcap #66: Fix intermittent start-up video desync

cspang1, 2018-09-03:
Occasionally, start-up will be desynced. Likely a race condition with a main memory access; need to deterministically time sync.

cspang1, 2019-03-22:
Per #63...

Relatively certain what's happening is that the worst-case scenario for the RX cog results in long count being loaded on Cog 0's side after the TX cog has started firing.

Likeliest solution is to pass `constant(GFX_BUFFER_SIZE << 16) | @gfx_buff` to the RX Cog and let it proceed to `waitpeq`ing internally.

cspang1, 2019-03-22:
May have fixed w/ commit 45524524218d5a4839a7c6a34e8fc2a364d77e4e, pending verification...

cspang1, 2019-03-24:
Looks like it may actually be caused by the write on the TX side to the `trans` address too soon...

cspang1, 2019-05-13:
Next debug step is to hardcode some long waits in critical sections to try and force the behavior.

cspang1, 2019-05-14:
This is 100% happening when TX is running BEFORE RX is, 0% vice versa - narrowing down cause via scratchpad and deterministic timing.

konimaru, 2019-05-14:
Couldn't you do something simple like a start-up handshake and then use said line(s) for normal transfer?

cspang1, 2019-05-14:
That's actually what I'm already doing! I'm using the vertical sync (`VS`) and TX/RX (`X`) line as a two-way handshake which accounts for either Prop starting up first: the GPU's RX cog waits for `VS` to go high, while the CPU's RX cog drives `V`S high and then immediately waits for `X` to go high. 

So if the CPU starts up first, it drives `VS` high and it waits for `X` to be driven high by the GPU. When the GPU eventually starts up, its `waitpeq` will immediately detect `VS` is high, and proceed to immediately drive `X` high, notifying the CPU to proceed as well.

If the GPU starts up first, it immediately starts waiting for `VS` to be high. When the CPU eventually starts up, it will drive `VS` high which causes the GPU to immediately drive `X` high, which notifies the CPU to proceed. 

This paradigm works the majority of the time, with video starting and displaying seamlessly when both uCs are started at (ostensibly) the same time.

The problem is that, occasionally and without any apparent deterministic pattern, the transfer will be very slightly desynced at startup - colors and patterns flashing around in incorrect places, but the basic elements are definitely there.

What is almost certainly happening, and I have a test planned for this evening to verify, is that the transfer itself is desyncing - what the RX cog thinks is the "start" of the data stream is actually within the stream elsewhere, off by some yet-unknown number of bits, bytes, words, or longs.

What I've been able to determine for certain so far is that this behavior *only* happens when the CPU is running *before* the GPU (TX cog running before RX cog). Even with this startup order enforced to purposely trigger the desync, it still doesn't happen every time (maybe 1/3rd of the time).

So now I'm in a position of trying to determine if this is:
1. A hardware instead of software problem - the breadboarded nature of the circuit is causing errant behavior in the transfer (unlikely, as it would then happen regardless of which uC started up first) or maybe the individual clocks of the uCs are aligning or misaligning or mistiming in just the right way to cause an off-by-one-cycle error (again, if so then why wouldn't this happen regardess of which uC started up first)
2. An edge case, where a worst-case hub access timing is causing the desync somehow (my original thoughts, as the difference between 8 and 23 execution cycles would fit like a glove for an occasional symptom like this - but after plotting out the worst-case execution times down to the clock cycle in Excel, I don't see anywhere that this would be happening)
3. Something I haven't considered yet, which is where I'm sitting now

The fact that I've narrowed it down to only happening when TX precedes RX should be a big hint, but the random nature even then of the behavior is eating my chili. My most recent attempt at rooting out the cause was trying to force it to happen 100% of the time, adding delays between critical sections, but that didn't bare any fruit.

[Here's the TX code with the handshake starting at line 86](https://github.com/cspang1/JCAP/blob/bugfix/video/startup/Dev/Software/tx.spin#L86), and [here's the RX code with the handshake starting at line 78](https://github.com/cspang1/JCAP/blob/bugfix/video/startup/Dev/Software/rx.spin#L78)

Somethign I may do tonight is remove the handshake code altogether, force TX to start before RX, and preclude that the handshake itself is the problem. If I can rule that out entirely, then that's another data point.

If you have any thoughts, they would be greatly appreciated. This issue has knocked me off of development for the past several months.

cspang1, 2019-05-14:
A new data point: forcing the RX cog to only accept a single transmission (followed by an infinite `jmp $`) shows me that the behavior is evident from the very first frame of data transmitted. I'm replacing one of the GPU rendering cogs with a serial transmission COG to sample the data that is in the video buffer to see whether the stored video data received from the CPU is errant.

cspang1, 2019-05-14:
Ok we're getting down to it now: 

11111111_00001111_11000011_00000011 <--- GOOD GFX DATA
11111111_10001111_11100011_10000011 <--- BAD GFX DATA
11111111_00000111_11000001_00000001 <--- BAD GFX DATA
01111111_10001111_11000011_00000011 <--- BAD GFX DATA

These are all samples of the very first `long` of the GFX buffer on the GPU. All of these values and more are being seen, when only the first is stored in that memory during nominal operation. The only obvious pattern here is that the only bits that are being corrupted are those which are adjacent to a bit toggle (i.e. the preceding or following bit is different than the current one). 

This therefore truly looks like the RX prop is somehow becoming de-synced despite the deterministic timing, and trying to read data off the wire at the instant the bits are changing; sometimes it gets the correct data, other times it gets the wrong next one.

Still digging for the cause.

cspang1, 2019-05-15:
Removing the handshake PASM entirely causes the problem to disappear. I'd like to say that this finally narrows the problem down to the ~10 lines of code which defines the handshake routine, but we'll see. Again, I've plotted out every instruction for the best and worst case timing scenarios down to the individual clock cycle and could not find anything that would cause this problem.

cspang1, 2019-05-15:
Finally narrowed down to the problem to, surprisingly enough, the TX side of the house. Only when the handshake code is present in the TX routine (with the RX-side handshake moved out to SPIN) does the problem occur. I'm sure there's a logical explanation regarding timings and whatnot, but for now the problem has been solved by moving all handshake code out to SPIN in the `main` functions. The next iteration will be to move the handshakes into the individual TX and RX module functions.

This issue has been (initially) satisfied by commit f8b7b4c3115bc056ab05216b7fd197fa726fc405

closed 2019-05-15 by cspang1