cspang1/jcap #71: Implement sprite sizing

cspang1, 2019-05-16:
Should be able to render sprites @ 8x8, 16x16, and maybe even 32x32 sizes, along with permutations of these sizes.

cspang1, 2019-05-30:
For wide horizontal sprites, will need to loop back around to palette load. Will also need to increase buffer overflow space by another 2 longs to compensate.

konimaru, 2019-05-30:
A bit OT: what’s the worst case timing for rendering one line? Do you have any numbers?

cspang1, 2019-05-30:
The worst case timing currently results when you have the maximum number of sprites per scanline end-to-end w/ no transparent pixels (just an 8x8 pixel completely filled square). That results in the maximum number of rendering instructions per scanline. The timing for this is obviously dependant on what the maximum sprites per scanline is set to. After last week's shakedown of the render code, I was able to hit 20+ sprites before artifacting started.

The upper bound though is easy to calculate: 6 render cogs @ 2 visible lines displayed per rendered one (2x upscaling), 31.778 us per scanline (by the VGA standard) - so 63.555 us per upscaled scanline, times 6   for the cogs sharing the load, and we get 381.33 us to render a scanline.

cspang1, 2019-05-30:
Need to loop back to [this line](https://github.com/cspang1/JCAP/blob/c272e3687cb5802c3a4eea1c16926adb485fc3b0/Dev/Software/vga_render.spin#L263) w/ an address increment. Will also potentially need to adjust offset/wrapping logic...

cspang1, 2019-05-31:
@konimaru want your opinion on this: by implementing sprite sizing, you're definitely saving cycles on the CPU-side of the house (only need to update on SAT entry for a 16x16 sprite instead of 4), as well as some on the GPU side (can skip some position checking instructions).

But, it makes the actual maximum sprites per scanline difficult to pin-down: you can probably fit 20+ 8x8 sprites on a scanline, or 10+ 16x16 sprites, but mixing the two or mixing in 16x8 and 8x16 makes dozens of permutations. My pseudo-OCD doesn't like this uncertainty, but the reduction in headache on the game dev side by not having to deal with every single 8x8 sprite individually really gives the feature some merit. Thoughts?

konimaru, 2019-06-01:
I see what you mean ... and you ask me with blood type A :)

That being said, if it's easy and has certain benefits (big sprites are cool) I'd go for it. As for **measuring** this, not sure if this holds true: Is it guaranteed that with a given N 8x8 sprites you can get effectively more of them per scanline when some of them are combined e.g. M 16x16 + P 8x8 (2*M+P>N)? If that's the case I'd simply state that you could have N base sprites and may get _more_ etc if and when. OTOH I wouldn't necessarily squeeze that into a formula.

If the above isn't true for all cases then we'd have to think of something else, maybe some form of optimistic limit (e.g. not 20 but just 16 sprites/scanline). I'll keep this issue in mind and think about it a bit more.

cspang1, 2019-06-02:
Thanks for the input - I'm definitely leaning heavily towards implementing them and letting the uncertainty be. Working out the implementation details has also revealed some further overall refinement, such as extending the scanline buffer at the beginning the same way I did to the end, allowing for rendering in non-displayed space without having to deal with calculating offsets for partially on-screen sprites.

cspang1, 2019-06-03:
Thinking with extended buffer method, will need index to reset to buffer origin `SprSzX` pixels "early", i.e. reduce `maxHor` by `SprSzX` prior to `cmpsub`. Remove shifting and offset logic, just use `cmpsub` result as origin for rendering sprite.

cspang1, 2019-06-12:
Life getting in the way of finally implementing this...

W/ buffer added to beginning, will need to compensate for 8-px wide sprites needing to start rendering in second half of extra buffer space.

cspang1, 2019-10-23:
Moved to a new state for a new job, finally everything's settled down enough to start working on this again...
[image]

konimaru, 2019-10-23:
Welcome back and best wishes!

cspang1, 2019-10-24:
Managed to implement a basic wide sprite, but horizontal/vertical/multidirectional mirroring is still left. Beginning to get a nagging suspicion that fully implementing this will take more of a toll on the max sprites per line metric than it's worth.

cspang1, 2019-10-25:
[Completely scrapped the janky horizontal wrapping offset calculation nonsense](https://github.com/cspang1/JCAP/commit/148e57ac9b6455805b2dbd23136f70c57316b043) and opted to simply extend the scanline buffer to the left of the screen and
1. Use 16 pixels left of screen for sprites sliding into the frame, and
2. Use sprite x position '0' as an invisible location, providing the benefit of an easy coordinate to hide unused sprites in the SAT

Adds only 4 longs of cog memory overhead for the additional buffer space, but many more removed (as well as resulting execution cycles) in removing the instructions for the offset calculation voodoo.

With this improvement, sprite sizing is looking (tentatively) much more reasonable.

cspang1, 2019-10-27:
Hard part done w/ horizontal wide + mirrored sprites successfully implemented via 15f7b4be0233b100009cebb433986ab3771f7a5c

cspang1, 2019-10-27:
Critical code section for refactoring:

```
sprites ' Load sprite attributes
        rdlong          curmt,  tmindx      ' Load sprite attributes from Main RAM
        mov             temp,   curmt       ' Copy sprite attributes to temp variable
        shr             temp,   #7          ' Shift vertical position to LSB
        and             temp,   #255 wz     ' Mask out vertical position, checking invisibility
        cmp             maxVVis, temp wc    ' Check sprite upper bound
        if_be jmp       #:skip              ' Skip sprite if invisible or out of bounds
        mov             spypos, temp        ' Store sprite vertical position        

        ' Check if sprite is on scanline vertically
        add             temp,   #system#SPR_SZ_S-1          ' Calculate sprite vertical position upper bound
        cmp             temp,   cursl wc                    ' Check sprite upper bound
        if_nc cmp       cursl,  spypos wc                   ' Check sprite lower bound
        if_nc jmp       #:cont                              ' Check sprite horizontally within scanline
        cmpsub          temp,   #system#MAX_SPR_VER_POS wc  ' Force wrap (carry if wrapped)
        if_c  cmpx      cursl,  temp wc                     ' Re-check bounds
        if_nc jmp       #:skip                              ' Skip sprite

        ' Calculate vertical sprite pixel palette offset
        mov             spyoff, #system#SPR_SZ_S-1  ' Copy sprite vertical size to sprite vertical offset
        sub             temp,   cursl               ' Subtract current scanline from sprite lower bound
        sub             spyoff, temp                ' Subtract vertical sprite position from vertical sprite size
:cont   if_nc mov       spyoff, cursl               ' Store current scanline into sprite offset
        if_nc sub       spyoff, spypos              ' Subtract vertical sprite position from sprite offset

        ' Check if sprite is within scanline horizontally
        mov             spxpos, curmt       ' Copy sprite attributes to temp variable
        shr             spxpos, #15         ' Shift horizontal position to LSB
        and             spxpos, #511 wz     ' Mask out horizontal position, checking invisivility
        cmp             maxHVis, spxpos wc  ' Check sprite upper bound
        if_be jmp       #:skip              ' Skip sprite if invisible or out of bounds
        test            curmt,  #2 wc       ' Check if sprite is wide
        if_c mov        widesp, #2          ' If so increment wide sprite index
        if_c mov        wmmod,  #8          ' And set sprite position modifier
        if_nc mov       widesp, #1          ' Otherwise only render one sprite palette line

        ' Retrieve sprite pixel palette line
        mov             temp,   curmt               ' Copy sprite attributes to temp variable
        shr             temp,   #24                 ' Align sprite pixel palette attribute to LSB
        shl             temp,   #5                  ' Calculate sprite pixel palette Main RAM location offset
        add             temp,   spptr               ' Calculate sprite pixel palette Main RAM base location
        test            curmt,  #8 wz               ' Check sprite mirrored vertically
        if_nz subs      spyoff, #system#SPR_SZ_S-1  ' If so calculate inverted offset...
        if_nz abs       spyoff, spyoff              ' And calculate final absolute offset
        shl             spyoff, #2                  ' Calculate vertical sprite pixel palette offset
        add             temp,   spyoff              ' Calculate final sprite pixel palette Main RAM location
        rdlong          curpt,  temp                ' Load sprite pixel palette line from Main RAM
```

cspang1, 2019-10-28:
Satisfied by commit 2708506cee6fffaded65836cb1a269189be6707c!

closed 2019-10-28 by cspang1