cspang1/jcap #75: minor improvements

konimaru, 2019-05-21:
```
slgen   'Calculate tile map line memory location
        mov             tmindx, cursl           ' Initialize tile map index
        shr             tmindx, #3              ' tmindx = floor(cursl/8)
        mov             temp,   tmindx          ' Store tile map index into temp variable

        mov             findx,  tmindx          ' Store tile map index into temp variable
        shl             temp,   #6              ' tmindx *= 64
        shl             findx,  #5              ' tmindx *= 32
        shl             tmindx, #4              ' tmindx *= 16
        add             findx,  temp            ' tmindx = tmindx(64+32)
        add             tmindx, findx           ' tmindx = tmindx(64+32+16)

        add             tmindx, tmptr           ' tmindx += tmptr + tmindx*112
        mov             initti, tmindx          ' Store initial row tile location
```
may become
```slgen   'Calculate tile map line memory location
        mov             tmindx, cursl           ' Initialize tile map index
        shr             tmindx, #3              ' tmindx = floor(cursl/8)
        mov             temp,   tmindx          ' Store tile map index into temp variable

        shl             tmindx, #3              ' x8
        sub             tmindx, temp            ' x7
        shl             tmindx, #4              ' x112

        add             tmindx, tmptr           ' tmindx = tmptr + (cursl/8)*112
        mov             initti, tmindx          ' Store initial row tile location
```

cspang1, 2019-05-21:
Good catch - intuitive math there!

If you're not bored of this code yet, the most critical points of optimization are this sequence of lines (executed 320 times/scanline):

```
        {{ HALF 1 PIXEL 1 }}
        mov             temp,   curpt           ' Load current palette tile into temp variable
        shr             temp,   #28             ' LSB align palette index
        add             temp,   cpindx          ' Calculate color palette offset
        rdbyte          curcp,  temp            ' Load color
        or              pxbuff, curcp           ' Store color
        ror             pxbuff, #8              ' Allocate space for next color
        shl             curpt,  #4              ' Shift palette tile left 4 bits

        djnz            pxindx, #:next1
        call            #tld

        {{ HALF 1 PIXEL 2 }}
:next1  mov             temp,   curpt           ' Load current palette tile into temp variable
           etc...
```

and this routine (executed 40 times/scanline):

```
tld     add             tmindx, #2              ' Increment pointer to tile in tile map
        djnz            remtil, #:next          ' Check if need to wrap to beginning
        if_z mov        tmindx, initti          ' If so wrap to beginning
:next   rdword          curmt,  tmindx          ' Load current map tile from Main RAM
        mov             cpindx, curmt           ' Store map tile into color palette index
        and             curmt,  #255            ' Isolate palette tile index of map tile
        shr             cpindx, #8              ' Isolate color palette index of map tile
        shl             cpindx, #4              ' cpindx *= 16
        add             cpindx, tcpptr          ' cpindx += tcpptr
        mov             tpindx, cursl           ' Initialize tile palette index
        and             tpindx, #7              ' tpindx %= 8
        shl             tpindx, #2              ' tpindx *= 4
        shl             curmt,  #5              ' tilePaletteIndex *= 32
        add             tpindx, curmt           ' tpindx += paletteTileIndex
        add             tpindx, tpptr           ' tpindx += tpptr
        rdlong          curpt,  tpindx          ' Load current palette tile from Main RAM
        mov             pxindx, #8
tld_ret ret
```

They're my current focus for any additional cycles I can squeeze out to give me some buffer when I add the few instructions needed for vertical scrolling.

konimaru, 2019-05-21:
The second part (**tld**) is just out of sync I noticed (11 instead of 10 insn). If you can accept _nasty hacks_ then I can offer this:
```
tld     add             tmindx, #2              ' Increment pointer to tile in tile map
        djnz            remtil, #:next          ' Check if need to wrap to beginning
        if_z mov        tmindx, initti          ' If so wrap to beginning
:next   rdword          curmt,  tmindx          ' Load current map tile from Main RAM
        mov             cpindx, curmt           ' Store map tile into color palette index
        and             curmt,  #255            ' Isolate palette tile index of map tile
        shr             cpindx, #8              ' Isolate color palette index of map tile
        shl             cpindx, #4              ' cpindx *= 16
        add             cpindx, tcpptr          ' cpindx += tcpptr
        mov             phsb,   cursl           ' Initialize tile palette index
        and             phsb,   #7              ' tpindx %= 8
        shl             phsb,   #2              ' tpindx *= 4
        shl             curmt,  #5              ' tilePaletteIndex *= 32
        add             phsb,   curmt           ' tpindx += paletteTileIndex
        rdlong          curpt,  phsb            ' Load current palette tile from Main RAM
        mov             pxindx, #8
tld_ret ret
```
**tpindx** is replaced by **phsb**, counter B runs in **logic.always** mode and **frqb** is loaded with **tpptr/2**. The **rdlong** effectively reads from **phsb**+2***frqb** so everything is good :)

cspang1, 2019-05-21:
You and your counter hacks haha. Let me wrap my head around this one!

cspang1, 2019-05-21:
Trivial improvement just because I don't like incrementing, storing, then decrementing:

```
  start_line_ := -1     ' Initialize first scanline index

...

        ' Get initial scanline and set next cogs via semaphore
:lock   lockset         semptr wc               ' Attempt to lock semaphore
        if_c  jmp       #:lock                  ' Re-attempt to lock semaphore
        rdlong          initsl, ilptr           ' Load initial scanline
        adds            initsl, #1              ' Increment initial scanline for next cog
        mov             cursl,  initsl          ' Initialize current scanline
        wrlong          initsl, ilptr           ' Write back next initial scanline
        lockclr         semptr                  ' Clear semaphore
        cogid           temp wz, nr             ' Check if this is the final cog to be initialized
        if_z  lockret   semptr                  ' Return lock handle if so
```

cspang1, 2019-05-21:
Thought: by checking `pxindx` and doing a `call` with the `tld` routine, we're adding (worst-case) 16x40=640x9.615 ns = 6.15 us. This repeated checking for tile turnover is only required for the first tile (the only one we might have to start rendering "partway" through).

How about extending the scanline buffer by another couple segments, with those segments being "offscreen" to the left, and starting rendering in them via a calculated offset.

cspang1, 2019-05-21:
So for a given scanline, regardless of how much shifting there is, the tile transition will always happen between the same pixels. So, we can dynamically load a `call` to the `tld` routine at a point between those pixels, replacing a `nop` originally there.

This would save... a lot of cycles. Call the `call`s and `nop`s a wash, we're removing all the `djnz`s for 280*8+40*4 and some others in the `tld` routine for around 2,560 cycles for a savings of 24.6 us - almost an entire scanline's worth of recovered render time.

konimaru, 2019-05-21:
Sounds worth doing then.

cspang1, 2019-05-21:
> Sounds worth doing then.

Definitely working on it - prefer to do a refactor that big when I'm in front of my equipment.

cspang1, 2019-05-21:
@konimaru can't seem to get your `phsb` trick working. I'm starting `ctrb` in logic.always mode with a simple:

```
mov ctrb, ctrcfg
...
ctrcfg long %0_11111_000_00000000_000000_000_000000
```

Tile graphics are absolutely jacked up. Thoughts?

konimaru, 2019-05-22:
Do you have your code somewhere? Otherwise I’ll prepare an example.

cspang1, 2019-05-22:
One moment, I'll throw it up on a branch

cspang1, 2019-05-22:
[Here you go](https://raw.githubusercontent.com/cspang1/JCAP/temp/video/maxspr/Dev/Software/vga_render.spin)

konimaru, 2019-05-22:
Well, we still need an initialized **tpptr** even if it's aliased to **frqb**. Change that:
```
        add             tpptr,  datptr  ' Calculate tile palette location
        rdlong          tpptr,  tpptr   ' Load tile palette location
```
to
```
        add             tpptr,  datptr  ' Calculate tile palette location
        rdlong          frqb,   tpptr   ' Load tile palette location
        shr             frqb,   #1      ' load from phsb+2*frqb
```
Also, `movi ctrb, #%0_11111_000` will do :)

cspang1, 2019-05-22:
Ah! Completely missed that, just not quite used to the `phsx` accumulation logic - works perfectly! Thanks for the guidance!

All of this and more satisfied by 4d855627642b216f5dca39a256b4fab2e51e03a8

konimaru, 2019-05-22:
```
tile    mov             pxbuf1, #0  ' Initialize first half-tile pixel buffer
        mov             pxbuf2, #0  ' Initialize second half-tile pixel buffer

        {{ HALF 1 PIXEL 1 }}
        mov             temp,   curpt   ' Load current palette tile into temp variable
        shr             temp,   #28     ' LSB align palette index
        add             temp,   cpindx  ' Calculate color palette offset
        rdbyte          curcp,  temp    ' Load color
        or              pxbuf1, curcp   ' Store color
        ror             pxbuf1, #8      ' Allocate space for next color
        shl             curpt,  #4      ' Shift palette tile left 4 bits
        nop
```
The first two **mov**s can be done in the {{ HALF ? PIXEL 1 }} segments, i.e.
```
        rdbyte          curcp,  temp    ' Load color
        mov             pxbuf1, curcp   ' Store color
        ...
        rdbyte          curcp,  temp    ' Load color
        mov             pxbuf2, curcp   ' Store color
```
Also, this is one insn shorter:
```
        ' Determine horizontal pixel location in tile
        mov             temp,   horpos  ' Store horizontal position into temp variable
        shl             temp,   #2      ' temp *= 4
        shl             curpt,  temp    ' Shift to first pixel
        shl             temp,   #1      ' temp *= 8
        and             temp,   #%111000
        mov             spypos, #lastpx ' Temporarily store possible tile load position of final pixel in imminent buffer
        sub             spypos, temp    ' Calculate offset
```
or - in case you don't need **horpos** in the future:
```
        ' Determine horizontal pixel location in tile
        shl             horpos, #2      ' *= 4
        shl             curpt,  horpos  ' Shift to first pixel
        shl             horpos, #1      ' *= 8
        and             horpos, #%111*8 ' limit
        mov             spypos, #lastpx ' Temporarily store possible tile load position of final pixel in imminent buffer
        sub             spypos, horpos  ' Calculate offset
```

cspang1, 2019-05-22:
Nice catch! I like the technique for doing the modulo operation with a shifted mask.

konimaru, 2019-05-22:
```
        {{ HALF 2 PIXEL 2 }}
        mov             temp,   curpt   ' Load current palette tile into temp variable
        shr             temp,   #28     ' LSB align palette index
        add             temp,   cpindx  ' Calculate color palette offset
        rdbyte          curcp,  temp    ' Load color
        or              pxbuf2, curcp   ' Store color
        ror             pxbuf2, #8      ' Allocate space for next color
        shl             curpt,  #4      ' Shift palette tile left 4 bits
        nop
```
(big grin)

Scratch the **nop** and use the ``` shl curpt,  #4``` as the patch location. IOW, it's either **shl** or the **call** to tld (which re-initialises **curpt** anyway so there is no need for the **shl**). Then we need to adjust the multiplier to x7 (relatively easy) and are in sync with hub most of the time.

cspang1, 2019-05-22:
Oh that's sneaky! Now that's another significant boost: 8 instr*320 pixels*9.615ns = 24.61 us!

cspang1, 2019-05-22:
Tentatively satisfied by 568b6b14ae293e290460506291bde5b4c482c0ec.

My adjustment of the multiplier to x7 feels a bit sloppy, but I think I've stared at this code enough for a few hours haha. Will verify functionality when I'm in front of my equipment.

konimaru, 2019-05-22:
x7
```
        ' Determine horizontal pixel location in tile
        mov	        spypos, horpos  ' *= 1
        and	        spypos, #%111*1 ' limit
		
        shl             horpos, #2      ' *= 4
        shl             curpt,  horpos  ' Shift to first pixel
        shl             horpos, #1      ' *= 8
        and             horpos, #%111*8 ' limit

        add             spypos, #lastpx ' lastpx + 1*(horpos&7)
        sub             spypos, horpos  ' lastpx - 7*(horpos&7)

trset   mov             0-0,    #0      ' Reset previous frame's tile load routine call
        movd            tiset,  spypos  ' Set next frame's reset
        movd            trset,  spypos  ' Set this frame's tile load routine call location
tiset   mov             0-0,    tldcall ' Set this frame's tile load 
```

konimaru, 2019-05-22:
Before **trset** we currently have 8 insn. I just came up with a 6 insn version.
```
                movd    frqa, horpos
                and     frqa, fmsk              ' %111 << 9
                shl     horpos, #2

                mov     phsa, #0
                shl     curpt, horpos
                add     frqa, phsa              ' (6+1)*(horpos&7)<<9

trset           mov     lastpx, insn_shl
                sub     tiset, frqa
                sub     trset, frqa
tiset           mov     lastpx, insn_call
```
Have fun wrapping your head around this. I'm going to bed now.

cspang1, 2019-05-22:
I knew you would slip a counter in somewhere haha

konimaru, 2019-05-22:
Just realised this doesn't quite work because **trset/tiset** are not reset. Oh well, 8 isn't too bad ... :)

konimaru, 2019-05-22:
... unles we have some idle hub slots ...

cspang1, 2019-05-22:
> ... unless we have some idle hub slots ...

How do you mean? As in holding for a hub access window alignment?

cspang1, 2019-05-23:
Pending identification of additional optimizations, above satisfied by 95bc26b6a31d2d65c15289795db731092fd11771

cspang1, 2019-05-23:
Random thought: would be nice to have a `abssub` function which evaluated to `|d-s|`

cspang1, 2019-05-23:
Implement @konimaru's [patch table method](https://github.com/cspang1/JCAP/commit/95bc26b6a31d2d65c15289795db731092fd11771#commitcomment-33646133).

cspang1, 2019-05-24:
Satisfied by 25a8912fb637a9e9e28e249ba21641a246ee260f

closed 2019-05-24 by cspang1