cspang1/jcap #77: inefficient hub transfer

konimaru, 2019-05-24:
```
        ' Write scanline buffer to video buffer in Main RAM
write   wrlong          slbuff+0, curvb         ' If so, write scanline buffer to Main RAM video buffer
        add             write,  d0              ' Increment scanline buffer memory location
        add             curvb,  #4              ' Increment video buffer memory location
        djnz            index,  #write          ' Repeat for all scanline segments
```
Something like [this](https://forums.parallax.com/discussion/comment/1108886/#Comment_1108886) would certainly be faster (and doesn't even use counters).

konimaru, 2019-05-24:
The cure loop contains 6 insn and we need 3 for setup (2x **movd**, 1x hub location with magic marker).

cspang1, 2019-05-24:
Wow that's some impressive bit magic! Implementing...

cspang1, 2019-05-26:
Satisfied by 927afe3df0f42e4ff23c15722e2fbdbc5b1e0691

Thanks again @konimaru!

closed 2019-05-26 by cspang1