cspang1/jcap #81: Implement parallax scrolling

cspang1, 2019-05-26:
Can either implement via NES-style "sprite marker" (last scanline of sprite 0 sets a flag), or use a more robust method by passing _N_ scanline indexes representing where change in scrolling takes place.

cspang1, 2019-05-28:
Or just read from a 30-long array containing individual horizontal positions for each row of tiles. Would be as simple as using the current scanline as the index for the array. Would introduce only a few instructions of overhead on the render side:

_cpu/gpu:_
```
parallax_arr byte 0[30]
```
_vga_render:_
```
plxptr = parallax_arr_base + cursl
rdlong horpos, plxptr
etc.
```

konimaru, 2019-05-28:
After last weeks session you should have lots of space/time :)

Why 30 in particular?

cspang1, 2019-05-28:
Haha thanks to you definitely. Screen is 30 tiles tall, so you could have every row of tiles parallaxing at different speeds. Doing it per individual scanline would also be more robust, allowing for the ability to create some really cool effects like heatwave shimmers and the like - which now that I think about it sounds pretty powerful.

cspang1, 2019-10-29:
With projected algorithm, 2 things are absolutely necessary:

1. Scanline 0 is initialized with in-bounds x/y position coordinates in `plx_pos`, and
2. Any unused parallax array entries have their scanline index (7:0) set higher than the highest scanline (240)

cspang1, 2019-10-31:
Successfully implemented multi-line parallax scrolling!

[image]

Only limitation currently is that you can't have parallaxing between lines <6 scanlines apart, but this should be resolved in the next iteration.

cspang1, 2019-11-01:
This godforsaken yet 100% worth-the-effort feature has been implemented as of 78842387112f13ae35a6649054fae9641b853826. Now, every single scanline can be individually parallaxed in both the X and Y directions. I've probably spent ~12 solid man-hours this week working and reworking the algorithm to get this functional. I've been thwarted by [a fat-fingered immediate value mishap](https://forums.parallax.com/discussion/comment/1480931#Comment_1480931), by a missing `wc` effect flag, and more incredibly by[ an errant effect flag check somehow not only being left in](https://github.com/cspang1/JCAP/commit/78842387112f13ae35a6649054fae9641b853826#diff-8e72ef7673bf00f0a1c7ceb6b8bb015aL356) after all of the past refactoring - but not even create [a completely baffling issue](https://github.com/cspang1/JCAP/commit/888871a3a573511c65c8ae4f5fdb323d330776aa) until dozens of commits later. What's even better about this one is I solved it thanks to [your extremely emphasized in red notes on flag clears caused by routine `call`s on the API wiki](https://lamestation.atlassian.net/wiki/spaces/PASM/pages/17924303/CALL) @konimaru - so thanks once again for the invaluable help :).

closed 2019-11-01 by cspang1